STMicroelectronics/STM32CubeU0 #3: warning: unused variable in stm32u0xx_hal_rtc.h

CanastraRF, 2024-06-10:
Hi 
In File stm32u0xx_hal_rtc.h are the following macros:

**#define __HAL_RTC_WRITEPROTECTION_DISABLE(__HANDLE__)
#define __HAL_RTC_WRITEPROTECTION_ENABLE(__HANDLE__)
#define __HAL_RTC_DAYLIGHT_SAVING_TIME_ADD1H(__HANDLE__, __BKP__)
#define __HAL_RTC_DAYLIGHT_SAVING_TIME_SUB1H(__HANDLE__, __BKP__)
#define __HAL_RTC_ALARMA_ENABLE(__HANDLE__)
#define __HAL_RTC_ALARMA_DISABLE(__HANDLE__)
#define __HAL_RTC_ALARMB_ENABLE(__HANDLE__)
#define __HAL_RTC_ALARMB_DISABLE(__HANDLE__)
#define __HAL_RTC_ALARM_ENABLE_IT(__HANDLE__, __INTERRUPT__)
#define __HAL_RTC_ALARM_DISABLE_IT(__HANDLE__, __INTERRUPT__)
#define __HAL_RTC_ALARM_GET_IT(__HANDLE__, __INTERRUPT__)
#define __HAL_RTC_ALARM_GET_IT_SOURCE(__HANDLE__, __INTERRUPT__)
#define __HAL_RTC_ALARM_GET_FLAG(__HANDLE__, __FLAG__)
#define __HAL_RTC_ALARM_CLEAR_FLAG(__HANDLE__, __FLAG__)**

that access the RTC directly and not with __HANDLE__.
So we get lot of warning: unused variable.

Please use (__HANDLE__)->Instance->... instead of RTC->... as in all other cubes (F0-F4, G0,G4,L0-F4) sources.

ALABSTM, 2024-06-14:
Hi @CanastraRF,

Would you please specify the compiler you are using, its version and the enabled flags?

Thanks,

CanastraRF, 2024-06-17:
Hi
We've code like the following:

static RTC_TimeTypeDef * RTC_HandleGet( void );
void RTC_AlarmAEnable(void);

RTC_TimeTypeDef	g_hRTC;

static RTC_TimeTypeDef * RTC_HandleGet( void )
{
	return &g_hRTC;
}

void RTC_AlarmAEnable(void)
{
	RTC_TimeTypeDef * const hRTC = RTC_HandleGet();
	__HAL_RTC_ALARMA_ENABLE( hRTC );
}

The warning is with GCC, ARMCLANG, VC++, PCLint, ...
There is no warning with STM32F0, STM32F1, STM32F2, STM32F3, STM32F4, STM32L0, STM32L1, STM32L4, STM32G0, STM32G4 Cube Files

What do you mean with enabled flags?

"-Wunused-variable" is a widely used warning.

ALABSTM, 2024-06-21:
ST Internal Reference: 184768

TOUNSTM, 2025-03-03:
Fixed in 959abee0d2618e4ee2021e3457620f2edf92886b

CanastraRF, 2025-03-03:
Hi
Thanks for the fix.
But this Fix is different to all other families STM32F0, STM32F1, STM32F2, STM32F3, STM32F4, STM32L0, STM32L1, STM32L4, STM32G0, STM32G4.
All other families has got code like this:

#define __HAL_RTC_WRITEPROTECTION_DISABLE(__HANDLE__)             \
                        do{                                       \
                            (__HANDLE__)->Instance->WPR = 0xCAU;   \
                            (__HANDLE__)->Instance->WPR = 0x53U;   \
                          } while(0U)

You add in STM32U0

#define __HAL_RTC_WRITEPROTECTION_DISABLE(__HANDLE__)             \
  do{                   \
    UNUSED(__HANDLE__); \
    RTC->WPR = 0xCAU;   \
    RTC->WPR = 0x53U;   \
  } while(0U)

Why the difference?

It would be helpful when all code are identical.
Please check it again