STMicroelectronics/STM32CubeU0 #5: Not helpful enum ERROR, SUCCESS  in cmsis_device_u0/Include/stm32u0xx.h

RetoFx, 2024-06-21:
Hi
In most STM32 families is ErrorStatus defined as followed:

typedef enum
{
  SUCCESS = 0U,
  ERROR = !SUCCESS
} ErrorStatus;

but in stm32u0xx.h is it reversed.
typedef enum
{
  ERROR = 0U,
  SUCCESS = !ERROR
} ErrorStatus;

This is not helpful for libraries.

Please fix this,

TOUNSTM, 2024-06-25:
ST Internal Reference: 184951

CanastraRF, 2024-06-26:
BTW:
ERROR and SUCCESS and ErrorStatus are widely-used names. Lots of libraries and programmer define this symbols.
But a project different definition results in big problem.
Example

Header1.h
#define ERROR 0xffffffffUL

Header2.h
typedef enum
{
SUCCESS = 0U,
ERROR = !SUCCESS
} ErrorStatus;

If both header are used in one translation units ... happy day

So please use unique symbol names like
STM32_ERROR
STM32_SUCCESS
....
But this breaks the compatibility with all exist code.
You can modify the library with

typedef enum
{
STM32_SUCCESS = 0U,
STM32_ERROR = !STM32_SUCCESS
} Stm32ErrorStatus;

#if defined(STM32_LEGACY)
 #if !defined(ERROR)
  #define ERROR STM32_ERROR
 #endif
 #if !defined(SUCCESS)
  #define SUCCESS STM32_SUCCESS
 #endif
 #if !defined(ErrorStatus)
  #define ErrorStatus Stm32ErrorStatus
 #endif
#endif

this would reduced lot of troubles.

TOUNSTM, 2024-07-17:
Hello @CanastraRF,

Thank you for your proposal. Our development teams acknowledged your proposal and its relevance. However, they have reported that there are currently no plans to support this enhancement.

Thank you for your comprehension and thank you again for your report.
With regards,

TOUNSTM, 2024-07-17:
Fixed in c661ffef8f49619e1c086079ed1149af089b372c

closed 2024-07-17 by TOUNSTM