STMicroelectronics/STM32CubeU0 #7: Missing #include <stddef.h> in sysmem.c

blackbird998, 2024-11-28:
Hi,
In `sysmem.c` files from `Projects/`  Templates/Examples `<stddef.h>` is not included directly (it provides `NULL` and `ptrdiff_t`).
In STM32CubeCTL_1.17.0 GNU-tools-for-STM32 arm-none-eabi implementation it is included indirectly by `<errno.h>` -> `<sys/errno.h>` -> `<sys/reent.h>` -> `<stddef.h>`, but this is not the case in e.g. LLVM-ET-Arm-19.1.1 implementation and can rise compile errors.
It could be fixed by directly including `<stddef.h>` in  `sysmem.c` files.

Kind regards,
Adam

ALABSTM, 2024-12-24:
Hi @blackbird998,

Thank you for this report. Did you experience compilation errors? If yes, would you please share a relevant snippet from the log to help us further analyse this point?

Thank you,

ALABSTM, 2024-12-24:
Hi @blackbird998,

We will check with our development teams if there is any plan to support the LLVM compiler and, if yes, how your request could be considered. We will keep you informed.

With regards,

TOUNSTM, 2024-12-25:
Hello @blackbird998,

Thank you for bringing this to our attention.  Unfortunately, our development team has decided not to make any changes to the current implementation. As we do not support the LLVM compiler, there are no plans to address this compatibility issue at this time. However, we may consider it for future development. 

Thank you again for your contribution.
Best Regards,

closed 2025-01-06 by ALABSTM