STMicroelectronics/STM32CubeU0 #9: Circular dependency and HAL_StatusTypeDef

blackbird998, 2025-02-24:
Hi,
When including any header from `STM32U0xx_HAL_Driver` and no other headers I get error `unknown type name 'HAL_StatusTypeDef'` which is defined in `stm32u0xx_hal_def.h`.
By checking how headers are included I found the following circular dependencies (`YYY` stands for module name, e.g. for `i2c` it would be `stm32u0xx_hal_i2c.h`):

- _**`stm32u0xx_hal_YYY.h`**_ includes `stm32u0xx_hal_def.h`
- `stm32u0xx_hal_def.h` includes `stm32u0xx.h`
- `stm32u0xx.h` includes `stm32u0xx_hal.h`
- `stm32u0xx_hal.h` includes `stm32u0xx_hal_conf.h`
- `stm32u0xx_hal_conf.h` includes _**`stm32u0xx_hal_YYY.h`**_

To avoid this error one of the following must be included first:
- `stm32u0xx.h` from `CMSIS`
- `stm32u0xx_hal.h` from `STM32U0xx_HAL_Driver`
- `main.h` from `Core` when using CubeMX

Is this by design?
This circular dependency could be fixed by removing:
```c
#if defined (USE_HAL_DRIVER)
#include "stm32u0xx_hal.h"
#endif /* USE_HAL_DRIVER */
```
from `stm32u0xx.h`. `stm32u0xx_hal.h` is already included in `main.h` and all `stm32u0xx_hal_YYY.c` files.

Kind regards,
Adam

KRASTM, 2025-02-27:
Hello @blackbird998,

Thank you for the report.

In fact, in the User manual **"Description of STM32xxx HAL and low-layer drivers"** of almost all STM32 series, you will find a chapter on **File inclusion model**, below an extract:

`The header of the common HAL driver file (stm32xxxx_hal.h) includes the common configurations for the whole
HAL library. It is the only header file that is included in the user sources and the HAL C sources files to be able to
use the HAL resources.`

[image]

So, it's recommended to follow this model, and for the circular dependency point, I'm going to share it with our team and keep you updated

With regards

KRASTM, 2025-03-28:
Hello @blackbird998,

I have checked with our team about the circular dependency point, but unfortunately there are no plans to remove this include from the CMSIS file.
Please, allow me to close the issue, thank you for your comprehension.

with regards,

closed 2025-03-28 by KRASTM