STMicroelectronics/STM32CubeU0 #10: RAM length in *.ld files

blackbird998, 2025-03-31:
Hi,
It seems that in most `*.ld` files RAM's `LENGTH`  accounts only for SRAM1 length (8KB for `U031xx` and 32KB for `U073xx`/`U083xx`, see [RM0503](https://www.st.com/resource/en/reference_manual/rm0503-stm32u0-series-advanced-armbased-32bit-mcus-stmicroelectronics.pdf#page=60)), while completely omitting SRAM2. According to [DS14581](https://www.st.com/resource/en/datasheet/stm32u031f4.pdf#page=9), SRAM2 is mapped in a way to offer "contiguous address space with SRAM1". Some of the `*.ld` files reflect that by defining RAM's `LENGTH` as `12K` or `40K`. Shouldn't RAM's `LENGTH` be a sum of SRAM1 and SRAM2 lengths? Or was it intentional? Alternatively could `RAM2` definition be added to the `MEMORY` section?

In addition, I think [DS14581](https://www.st.com/resource/en/datasheet/stm32u031f4.pdf#page=9) contains typo in SRAM2 address: `0x2000 8000`. 8KB of SRAM1 start at `0x2000 0000` and end at `0x2000 1FFF`, so SRAM1 should start at `0x2000 2000`. The `0x2000 8000` address would make sense for  `U073xx`/`U083xx` devices with 32KB of SRAM1 with end at `0x2000 7FFF`. It was probably copied from  DS14463 or DS14548.

Kind regards
Adam

ALABSTM, 2025-04-04:
Hi @blackbird998,

This is intentional, as you guessed. Actually, the linker files are provided under the `./Drivers/CMSIS/Device/ST/STM32U0xx/Source/Templates` [folder](https://github.com/STMicroelectronics/cmsis-device-u0/tree/f24a0240b77b0348826f8aeaf4860eb13fb74b4c/Source/Templates/gcc/linker). This means they are **templates** intended to be **customized** by the user based on their application's needs.

On the other hand, those found under `./Projects` have already been customized according to the example, the application, or the demo's needs. In case **SRAM2** is to be used, the RAM's length in the linker file has been modified accordingly.

I hope this clarifies the point. Do not hesitate otherwise.

With regards,

ALABSTM, 2025-04-28:
Hi @blackbird998,

Hope you're doing well and that my previous comment clarified the point and answered your question. Please allow me to close this issue. Thank you for your comprehension.

With regards

closed 2025-04-28 by ALABSTM